Tavish9/any4lerobot #3: Droid, language table dataset

lzl2040, 2025-02-23:
The size of droid and language table are greater than 1TB. How long will it take to convert these two datasets? Will there be the problem of out of memeory?

Tavish9, 2025-02-23:
hi, @lzl2040 

Based on the current progress, processing the droid dataset will take at least a week, and uploading it to Hugging Face will require a few additional days. (Droid is approximately 10 times larger than Bridge, and the processing + upload time for Bridge is around 2 days). Therefore, using the current script to process both the droid and language table datasets is not feasible.

We are currently working on a multi-processed conversion approach, which will be used to process these two large datasets in the future.

Aside from these two larger datasets, all other datasets have already been uploaded to [Hugging Face](https://huggingface.co/IPEC-COMMUNITY) and are available for direct download. (kuka and bc_z are on their way:)

lzl2040, 2025-02-24:
Thanks for your reply! When will you release the multi-processing code?

Tavish9, 2025-02-24:
The code will be released very soon! It’s currently being tested. Stay tuned

closed 2025-03-22 by lzl2040