Tavish9/any4lerobot #4: Transform my custom rlds dataset to lerobot.

islexu, 2025-02-27:
Thanks for the work. I saw you also use examples/libero/convert_libero_data_to_lerobot.py in [https://github.com/Physical-Intelligence/openpi/blob/main/examples/libero/convert_libero_data_to_lerobot.py](url), and the result is [https://huggingface.co/datasets/IPEC-COMMUNITY/libero_spatial_no_noops_image_lerobot](url), if I am not mistaken.

But this structure misses the video file and add the pics to the data file. Which can not be visualized in hugging face visualizer. So  I tried your method to have a structure like

'''
        ├── data
        │   ├── chunk-000
        │   │   ├── episode_000000.parquet
        │   │   ├── episode_000001.parquet
        │   │   ├── episode_000002.parquet
        │   │   └── ...
        │   ├── chunk-001
        │   │   ├── episode_001000.parquet
        │   │   ├── episode_001001.parquet
        │   │   ├── episode_001002.parquet
        │   │   └── ...
        │   └── ...
        ├── meta
        │   ├── episodes.jsonl
        │   ├── info.json
        │   ├── stats.json
        │   └── tasks.jsonl
        └── videos
            ├── chunk-000
            │   ├── observation.images.laptop
            │   │   ├── episode_000000.mp4
            │   │   ├── episode_000001.mp4
            │   │   ├── episode_000002.mp4
            │   │   └── ...
            │   ├── observation.images.phone
            │   │   ├── episode_000000.mp4
            │   │   ├── episode_000001.mp4
            │   │   ├── episode_000002.mp4
            │   │   └── ...
            ├── chunk-001
            └── ...

'''

But when i clone the repo and flow the tips, it turned out that
```
(lerobot) f50046112@m2408023435:~/lerobot/openx2lerobot$ bash convert.sh
2025-02-27 19:17:45.594776: I tensorflow/core/util/port.cc:153] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2025-02-27 19:17:45.602813: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:477] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
E0000 00:00:1740680265.612177 1025416 cuda_dnn.cc:8310] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
E0000 00:00:1740680265.615160 1025416 cuda_blas.cc:1418] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2025-02-27 19:17:45.625602: I tensorflow/core/platform/cpu_feature_guard.cc:210] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX512F AVX512_VNNI AVX512_BF16 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
I0000 00:00:1740680268.674980 1025416 gpu_device.cc:2022] Created device /job:localhost/replica:0/task:0/device:GPU:0 with 1051 MB memory:  -> device: 0, name: NVIDIA H100 NVL, pci bus id: 0000:c1:00.0, compute capability: 9.0
2025-02-27 19:17:48.928101: I tensorflow/core/kernels/data/tf_record_dataset_op.cc:376] The default buffer size is 262144, which is overridden by the user specified `buffer_size` of 8388608
Traceback (most recent call last):
  File "/home/f50046112/lerobot/openx2lerobot/openx_rlds.py", line 325, in <module>
    main()
  File "/home/f50046112/lerobot/openx2lerobot/openx_rlds.py", line 321, in main
    create_lerobot_dataset(**vars(args))
  File "/home/f50046112/lerobot/openx2lerobot/openx_rlds.py", line 229, in create_lerobot_dataset
    save_as_lerobot_dataset(
  File "/home/f50046112/lerobot/openx2lerobot/openx_rlds.py", line 157, in save_as_lerobot_dataset
    lerobot_dataset.save_episode(task=task_str)
  File "/home/f50046112/lerobot/lerobot/common/datasets/lerobot_dataset.py", line 799, in save_episode
    self._save_episode_table(episode_buffer, episode_index)
  File "/home/f50046112/lerobot/lerobot/common/datasets/lerobot_dataset.py", line 815, in _save_episode_table
    ep_dataset = datasets.Dataset.from_dict(episode_dict, features=self.hf_features, split="train")
  File "/home/f50046112/miniconda3/envs/lerobot/lib/python3.10/site-packages/datasets/arrow_dataset.py", line 945, in from_dict
    pa_table = InMemoryTable.from_pydict(mapping=mapping)
  File "/home/f50046112/miniconda3/envs/lerobot/lib/python3.10/site-packages/datasets/table.py", line 757, in from_pydict
    return cls(pa.Table.from_pydict(*args, **kwargs))
  File "pyarrow/table.pxi", line 1968, in pyarrow.lib._Tabular.from_pydict
  File "pyarrow/table.pxi", line 6337, in pyarrow.lib._from_pydict
[... 154 lines cut ...]
                                    },
                                    "description": "Reward if provided, 1 on final step for demos."
                                }
                            }
                        }
                    },
                    "length": "-1"
                }
            },
            "episode_metadata": {
                "pythonClassName": "tensorflow_datasets.core.features.features_dict.FeaturesDict",
                "featuresDict": {
                    "features": {
                        "file_path": {
                            "pythonClassName": "tensorflow_datasets.core.features.text_feature.Text",
                            "text": {},
                            "description": "Path to the original data file."
                        }
                    }
                }
            }
        }
    }
}
```

islexu, 2025-03-03:
Thank you for the instant response, I did as you said
1. add
```
    "sim_gen_ur3e_simple_grasp400_dataset": {
        "image_obs_keys": {"primary": "environment_image", "secondary": None, "wrist": "wrist_image"},
        "depth_obs_keys": {"primary": None, "secondary": None, "wrist": None},
        "state_obs_keys": ["state"],
        "state_encoding": StateEncoding.POS_EULER,  # state 包含 6 个关节角度和其他信息
        "action_encoding": ActionEncoding.EEF_POS,  # action 包含 3 个末端执行器位置变化和 1 个夹爪命令
        "control_frequency": 20,
        "robot_type": "UR3e",
    }, 
```
in  **config.py**

2. add
```
def simgen_dataset_transform(trajectory: Dict[str, Any]) -> Dict[str, Any]:
    # flip wrist_image from bgr to rgb
    trajectory["observation"]["environment_image"] = trajectory["observation"]["environment_image"][..., ::-1]

    trajectory["observation"]["wrist_image"] = trajectory["observation"]["wrist_image"][..., ::-1]
    trajectory["observation"]["state"] = trajectory["observation"]["state"]
    #fxu to do change 10 dim

    trajectory["action"] = trajectory["action"]

    trajectory["language_instruction"] = trajectory["language_instruction"]
    return trajectory
```
and

```
OXE_STANDARDIZATION_TRANSFORMS = {
...
"sim_gen_ur3e_simple_grasp400_dataset": simgen_dataset_transform,
}
```
in **transform.py**

3. change function  generate_features_from_raw to
```
def generate_features_from_raw(builder: tfds.core.DatasetBuilder, use_videos: bool = True):
    dataset_name = Path(builder.data_dir).parent.name

    state_names = [f"motor_{i}" for i in range(10)] #fxu 8 -10
    if dataset_name in OXE_DATASET_CONFIGS:
        state_encoding = OXE_DATASET_CONFIGS[dataset_name]["state_encoding"]
        if state_encoding == StateEncoding.POS_EULER:
            state_names = ["x", "y", "z", "roll", "pitch", "yaw", "pad", "gripper"]
            if "libero" in dataset_name:
                state_names = ["x", "y", "z", "roll", "pitch", "yaw", "gripper", "gripper"]  # 2D gripper state
            if "simgen" in dataset_name:
                state_names = ["x", "y", "z", "j1", "j2", "j3", "j4", "j5", "j6", "gripper"]
        elif state_encoding == StateEncoding.POS_QUAT:
            state_names = ["x", "y", "z", "rx", "ry", "rz", "rw", "gripper"]
        elif state_encoding == StateEncoding.JOINT:
            state_names = [f"motor_{i}" for i in range(7)] + ["gripper"]
            state_obs_keys = OXE_DATASET_CONFIGS[dataset_name]["state_obs_keys"]
            pad_count = state_obs_keys[:-1].count(None)
            state_names[-pad_count - 1 : -1] = ["pad"] * pad_count
            state_names[-1] = "pad" if state_obs_keys[-1] is None else state_names[-1]

    action_names = [f"motor_{i}" for i in range(4)] #fxu 8-4
    if dataset_name in OXE_DATASET_CONFIGS:
        action_encoding = OXE_DATASET_CONFIGS[dataset_name]["action_encoding"]
        if action_encoding == ActionEncoding.EEF_POS:
            # fxu action_names = ["x", "y", "z", "roll", "pitch", "yaw", "gripper"] to
            action_names = ["x", "y", "z", "gripper"]
        elif action_encoding == ActionEncoding.JOINT_POS:
[... 51 lines cut ...]
  File "/home/f50046112/lerobot/lerobot/common/datasets/lerobot_dataset.py", line 815, in _save_episode_table
    ep_dataset = datasets.Dataset.from_dict(episode_dict, features=self.hf_features, split="train")
  File "/home/f50046112/miniconda3/envs/lerobot/lib/python3.10/site-packages/datasets/arrow_dataset.py", line 945, in from_dict
    pa_table = InMemoryTable.from_pydict(mapping=mapping)
  File "/home/f50046112/miniconda3/envs/lerobot/lib/python3.10/site-packages/datasets/table.py", line 757, in from_pydict
    return cls(pa.Table.from_pydict(*args, **kwargs))
  File "pyarrow/table.pxi", line 1968, in pyarrow.lib._Tabular.from_pydict
  File "pyarrow/table.pxi", line 6337, in pyarrow.lib._from_pydict
  File "pyarrow/array.pxi", line 402, in pyarrow.lib.asarray
  File "pyarrow/array.pxi", line 252, in pyarrow.lib.array
  File "pyarrow/array.pxi", line 114, in pyarrow.lib._handle_arrow_array_protocol
  File "/home/f50046112/miniconda3/envs/lerobot/lib/python3.10/site-packages/datasets/arrow_writer.py", line 243, in __arrow_array__
    out = cast_array_to_feature(
  File "/home/f50046112/miniconda3/envs/lerobot/lib/python3.10/site-packages/datasets/table.py", line 1797, in wrapper
    return func(array, *args, **kwargs)
  File "/home/f50046112/miniconda3/envs/lerobot/lib/python3.10/site-packages/datasets/table.py", line 2108, in cast_array_to_feature
    raise TypeError(f"Couldn't cast array of type\n{_short_str(array.type)}\nto\n{_short_str(feature)}")
TypeError: Couldn't cast array of type
list<item: float>
to
Sequence(feature=Value(dtype='float32', id=None), length=8, id=None)

```

Maybe you have some clue about this, thank you in advance!

Tavish9, 2025-03-04:
Hi, as I said before, if you want to keep all the original state and action info, like your case (state=10, action=4), you need to add a custom encoding for both state and action.

However, in your reply, you kept all the original info but used the pre-defined StateEncoding and ActionEncoding, which causes the error.

Two tips:
- change your state&action format to fit the pre-defined StateEncoding and ActionEncoding (8-dim or 7-dim, pad if needed).
- or, add your custom StateEncoding and ActionEncoding, and then fix the parsing logic in [generate_features_from_raw](https://github.com/Tavish9/openx2lerobot/blob/main/openx_rlds.py#L83)

closed 2025-03-12 by Tavish9