Tavish9/any4lerobot #5: There are many blank tasks in the Bridge dataset. Is this normal?

venchi012, 2025-03-07:
[image]

Tavish9, 2025-03-07:
Yes, this is consistent with the original dataset. Please visit the [official website](https://rail-berkeley.github.io/bridgedata/) for more details.

closed 2025-03-08 by Tavish9

Saberlve, 2025-12-31:
Hello,

I am currently converting the [Bridge v2](https://huggingface.co/datasets/IPEC-COMMUNITY/bridge_orig_lerobot) dataset—from its original LeRobot v2.0 format—to the v2.1 format. However, the conversion process becomes extremely slow and consistently stalls at approximately 63% completion, after which an error is raised.

Have you encountered this issue before, or do you have any suggestions for troubleshooting it?