Tavish9/any4lerobot #6: TypeError: LeRobotDataset.save_episode() got an unexpected keyword argument 'keep_images'

islexu, 2025-03-11:
Thanks for your work again to keep up the openpi updates!

for the new scripts, I got the error:

(lerobot) f50046112@m2408023435:~/lerobot/openx2lerobot$ bash convert.sh
2025-03-11 16:31:24.230511: I tensorflow/core/util/port.cc:153] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2025-03-11 16:31:24.238800: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:477] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
E0000 00:00:1741707084.248109  879919 cuda_dnn.cc:8310] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
E0000 00:00:1741707084.250901  879919 cuda_blas.cc:1418] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2025-03-11 16:31:24.261187: I tensorflow/core/platform/cpu_feature_guard.cc:210] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX512F AVX512_VNNI AVX512_BF16 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
I0000 00:00:1741707087.301341  879919 gpu_device.cc:2022] Created device /job:localhost/replica:0/task:0/device:GPU:0 with 93011 MB memory:  -> device: 0, name: NVIDIA H100 NVL, pci bus id: 0000:c1:00.0, compute capability: 9.0
2025-03-11 16:31:28.474589: I tensorflow/core/kernels/data/tf_record_dataset_op.cc:376] The default buffer size is 262144, which is overridden by the user specified `buffer_size` of 8388608
Traceback (most recent call last):
  File "/home/f50046112/lerobot/openx2lerobot/openx_rlds.py", line 305, in <module>
    main()
  File "/home/f50046112/lerobot/openx2lerobot/openx_rlds.py", line 301, in main
    create_lerobot_dataset(**vars(args))
  File "/home/f50046112/lerobot/openx2lerobot/openx_rlds.py", line 221, in create_lerobot_dataset
    save_as_lerobot_dataset(lerobot_dataset, raw_dataset, keep_images=keep_images)
  File "/home/f50046112/lerobot/openx2lerobot/openx_rlds.py", line 157, in save_as_lerobot_dataset
    lerobot_dataset.save_episode(keep_images=kwargs.get("keep_images", False))
TypeError: LeRobotDataset.save_episode() got an unexpected keyword argument 'keep_images'

Do you have any idea about the underlying reason? Thanks in advance!

Tavish9, 2025-03-12:
Hi, please check the [README.md](https://github.com/Tavish9/openx2lerobot?tab=readme-ov-file#get-started) for details. 

It's time-consuming to delete images during creating dataset, so I recommend keeping images during conversion and delete them later.

closed 2025-03-13 by Tavish9

Tavish9, 2026-01-05:
@haofuly, yes `keep_images` is useless now, I will update the code later.

If you are using `v3.0`, you don't have to handle the deletion as it's done by multi-processing. Otherwise, you can delete all the images after the whole converting process. You have to save images if the raw dataset is stored as rlds format.