Tavish9/any4lerobot #8: Download problem for bc_z_lerobot

rakhimovv, 2025-03-13:
Hello,  

Thanks for sharing the IPEC-COMMUNITY/bc_z_lerobot dataset with the community!  

I'm encountering download failures when trying to access the dataset.  

It seems to be a permissions issue, as the system returns the following error:  
```
Failed to load dataset, retrying... Error: 403 Forbidden: None.
Cannot access content at: https://cdn-lfs-us-1.hf.co/repos/51/9b/519bd9a23dbf805ec096e1cbd20f9e139809e93516f4e3b59a3c1d6d4d5b4fa1/240d581c6d67fde3b3e4f42a8482762ff991b6b0800bfea8b7489bf426b388a1?response-content-disposition=inline%3B+filename*%3DUTF-8%27%27episode_000000.parquet%3B+filename%3D%22episode_000000.parquet%22%3B&Expires=1741879430&Policy=eyJTdGF0ZW1lbnQiOlt7IkNvbmRpdGlvbiI6eyJEYXRlTGVzc1RoYW4iOnsiQVdTOkVwb2NoVGltZSI6MTc0MTg3OTQzMH19LCJSZXNvdXJjZSI6Imh0dHBzOi8vY2RuLWxmcy11cy0xLmhmLmNvL3JlcG9zLzUxLzliLzUxOWJkOWEyM2RiZjgwNWVjMDk2ZTFjYmQyMGY5ZTEzOTgwOWU5MzUxNmY0ZTNiNTlhM2MxZDZkNGQ1YjRmYTEvMjQwZDU4MWM2ZDY3ZmRlM2IzZTRmNDJhODQ4Mjc2MmZmOTkxYjZiMDgwMGJmZWE4Yjc0ODliZjQyNmIzODhhMT9yZXNwb25zZS1jb250ZW50LWRpc3Bvc2l0aW9uPSoifV19&Signature=Ez1hfPxqkmlf4zOWL6Tqq-g6K3xlbjCGqOVtVAJwxIWL6fXaJP-QokYXI2s3Nk0bhUexaIDlSktgs2QhobKTgy80Wqp6kK1m1P6UL87u4Kdbaj0ikP1wMf5CyBRwPpvPIyNYudymUqtBWWXRdMh6vtRYUIB97VqL4xBbNgDnvj16HKFI0NiBtYtrTpXGRM2M~YSWQ6cP7IcduF2IoPt1HoO3bAvVC4gepstwNiM0EJyyoDGQgklDveeIiOYOeZEg4tuJ3TjLzVNK96aAYFgjPZz6QUo~qKLdWAaDbJwhmTOVnK4nc-PDEGI5cZnicH1PNgkf-MTjntDhVSQ06rPq4w__&Key-Pair-Id=K24J24Z295AEI9.
Make sure your token has the correct permissions.
<?xml version="1.0" encoding="UTF-8"?>
<Error><Code>AccessDenied</Code><Message>Access Denied</Message></Error>

```
I haven't encountered this issue with several other datasets.  

Could you please help resolve this? Thanks for your assistance!

Tavish9, 2025-03-13:
Hi, I think it might be because the number of files is too large, and HuggingFace has rate-limited your IP. You can try again after some time, or consider using another mirror service to download the files.

rakhimovv, 2025-03-13:
I will try, thanks, will report later

rakhimovv, 2025-03-14:
Tried several more times, no change.

I see a similar issue here https://huggingface.co/datasets/IPEC-COMMUNITY/bc_z_lerobot/discussions/3

Somehow the problem is with this particular dataset

Tavish9, 2025-03-15:
okay, maybe i need to upload it again. so weird

Tavish9, 2025-03-15:
I have finished the upload. Could you please try again and share the feedback? Thanks!

rakhimovv, 2025-03-16:
Downloaded successfully, thank you a lot, @Tavish9

closed 2025-03-16 by rakhimovv