Tavish9/any4lerobot #9: Download problem for language_table_lerobot

rakhimovv, 2025-03-13:
Hello,

Thanks a lot for sharing the IPEC-COMMUNITY/language_table_lerobot  dataset with the community!

I'm encountering download failures when trying to access the dataset.

```
Fetching 45153 files:  99%|█████████▉| 44819/45153 [00:37<00:00, 1097.01it/s]
Fetching 45153 files: 100%|█████████▉| 44936/45153 [00:37<00:00, 1117.33it/s]
Fetching 45153 files: 100%|█████████▉| 45064/45153 [00:37<00:00, 1160.51it/s]
Fetching 45153 files: 100%|██████████| 45153/45153 [00:37<00:00, 1207.28it/s]
Failed to load dataset, retrying... Error: index 563098 is out of bounds for axis 0 with size 563081
```

I run several times, the same problem arises

Tavish9, 2025-03-13:
please wait at least 1 week for this dataset, as the number of data files is the largest among openx. besides, i am uploading it now😂it is not completed yet

rakhimovv, 2025-03-13:
Ah, I see! 😅 thanks

Tavish9, 2025-03-15:
The speed of uploading language_table is acceptable. You can download it now🥳

closed 2025-03-15 by Tavish9