Tavish9/any4lerobot #11: lerobot2openx

Elycyx, 2025-03-29:
Thank you for your valuable work.

Have you considered developing a reverse conversion tool (lerobot2openx)? Many models only support datasets in the OXE format rather than the Lerobot format, but the Lerobot data collection pipeline is indeed more convenient.

Tavish9, 2025-03-29:
Great suggestion. I might implement lerobot2openx in the near future, as it aligns with the community's needs. Follow tightly and stay tuned!

Tavish9, 2025-04-07:
Would you prefer a new file or a new repo for lerobot2openx? 🙂

Tavish9, 2025-04-11:
I have restructured the repo, lerobot2openx would come soon.

Elycyx, 2025-04-11:
Thank you very much for your work — really looking forward to this feature!

Elycyx, 2025-04-11:
By the way, have you considered developing a tool for modifying datasets in the lerobot format? For example, functionalities like merging or splitting datasets, or modifying specific components such as RGBs or actions. I believe this would be very meaningful for the development of the lerobot community.

Elycyx, 2025-04-12:
I have a question that might not be directly related to this topic: In LeRobot, is it necessary to manually normalize the demonstration data during data collection, or will normalization of the dataset be done automatically during training?

Tavish9, 2025-04-12:
You don't need to normalize the raw trajectory when collecting raw data. You can do it by implementing a transformation in lerobot during training.

Elycyx, 2025-04-12:
> You don't need to normalize the raw trajectory when collecting raw data. You can do it by implementing a transformation in lerobot during training.

I got it, thanks a lot 🤗

closed 2025-04-23 by Tavish9