Tavish9/any4lerobot #24: How to merge datasets when facing the situation when state and action are not in  the same dimension

hairuoliu1, 2025-05-11:
### Question

In some cases, we may use different dims for action and state separately, hoping for support for this.

Tavish9, 2025-05-12:
Currently, the merging util supports the functionality you mentioned, see [README.md](https://github.com/Tavish9/any4lerobot/tree/main/utils/dataset_merging#%E5%85%AD%E5%8A%9F%E8%83%BD%E7%BB%86%E8%8A%82)

hairuoliu1, 2025-05-12:
Emm, I mean different dims for action and state separately, but the script just pads them with the same `max_dim`.

Tavish9, 2025-05-12:
You mean you want `state_max_dim` and `action_max_dim`, right?

hairuoliu1, 2025-05-12:
Yeah, and I can pull a request as it is easy.

Tavish9, 2025-05-12:
Many thanks if you are free to contribute!

TangGuohh, 2025-05-20:
> Yeah, and I can pull a request as it is easy.

Thank you for your attention. This feature will likely be supported around May 21st.

PR #30 (2025-05-21): Split the original single max_dim parameter into two separate parameters: state_max_dim and action_max_dim
# What does this PR do?

Fixes #24 

## Before submitting
- [ ] This PR fixes a typo or improves the docs (you can dismiss the other checks if that's the case).
- [x] Was this discussed/approved via a GitHub issue? Please add a link to it if that's the case.
- [x] Did you make sure to update the documentation with your changes?
- [ ] Did you write any new necessary tests?

## Who can review?

Anyone in the community is free to review the PR once the tests have passed. Feel free to tag
mem [... cut]
closed 2025-05-21 by Tavish9