Tavish9/any4lerobot #25: Not able to merge "image" mode lerobot datasets

hairuoliu1, 2025-05-11:
### System Info

https://github.com/Physical-Intelligence/openpi/blob/main/examples/aloha_real/convert_aloha_data_to_lerobot.py

### Reproduction

1. Collect raw h5 data
2. Use the script above to convert raw data to lerobot data
3. Use the merge script to merge multiple datasets
4. When loading data after merging, it occurred "can't infer dtype of dict".

I found that the codebase will lose information during the merging process for the image mode lerobot dataset, leading to problems when trying to read image data correctly.

### Expected behavior

Traceback (most recent call last):
File "/home/qiaojun/flexiv\_pi0-dev/scripts/compute\_norm\_stats.py", line 75, in <module>
tyro.cli(main)
File "/home/qiaojun/flexiv\_pi0-dev/.venv/lib/python3.11/site-packages/tyro/\_cli.py", line 189, in cli
return run\_with\_args\_from\_cli()
^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/qiaojun/flexiv\_pi0-dev/scripts/compute\_norm\_stats.py", line 62, in main
for batch in tqdm.tqdm(data\_loader, total=num\_frames, desc="Computing stats"):
File "/home/qiaojun/flexiv\_pi0-dev/.venv/lib/python3.11/site-packages/tqdm/std.py", line 1181, in **iter**
for obj in iterable:
File "/home/qiaojun/flexiv\_pi0-dev/src/openpi/training/data\_loader.py", line 257, in **iter**
batch = next(data\_iter)
^^^^^^^^^^^^^^^
File "/home/qiaojun/flexiv\_pi0-dev/.venv/lib/python3.11/site-packages/torch/utils/data/dataloader.py", line 708, in **next**
data = self.\_next\_data()
^^^^^^^^^^^^^^^^^
File "/home/qiaojun/flexiv\_pi0-dev/.venv/lib/python3.11/site-packages/torch/utils/data/dataloader.py", line 1480, in \_next\_data
return self.\_process\_data(data)
^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/qiaojun/flexiv\_pi0-dev/.venv/lib/python3.11/site-packages/torch/utils/data/dataloader.py", line 1505, in \_process\_data
data.reraise()
File "/home/qiaojun/flexiv\_pi0-dev/.venv/lib/python3.11/site-packages/torch/\_utils.py", line 733, in reraise
raise exception
RuntimeError: Caught RuntimeError in DataLoader worker process 0.
Original Traceback (most recent call last):
File "/home/qiaojun/flexiv\_pi0-dev/.venv/lib/python3.11/site-packages/torch/utils/data/\_utils/worker.py", line 349, in \_worker\_loop
data = fetcher.fetch(index)  # type: ignore\[possibly-undefined]
^^^^^^^^^^^^^^^^^^^^
File "/home/qiaojun/flexiv\_pi0-dev/.venv/lib/python3.11/site-packages/torch/utils/data/\_utils/fetch.py", line 52, in fetch
data = \[self.dataset\[idx] for idx in possibly\_batched\_index]
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/qiaojun/flexiv\_pi0-dev/.venv/lib/python3.11/site-packages/torch/utils/data/\_utils/fetch.py", line 52, in <listcomp>
data = \[self.dataset\[idx] for idx in possibly\_batched\_index]
\~\~\~\~\~\~\~\~\~\~\~\~^^^^^
File "/home/qiaojun/flexiv\_pi0-dev/src/openpi/training/data\_loader.py", line 47, in **getitem**
return self.\_transform(self.\_dataset\[index])
\~\~\~\~\~\~\~\~\~\~\~\~\~^^^^^^^
File "/home/qiaojun/flexiv\_pi0-dev/src/openpi/training/data\_loader.py", line 47, in **getitem**
return self.\_transform(self.\_dataset\[index])
\~\~\~\~\~\~\~\~\~\~\~\~\~^^^^^^^
File "/home/qiaojun/flexiv\_pi0-dev/.venv/lib/python3.11/site-packages/lerobot/common/datasets/lerobot\_dataset.py", line 652, in **getitem**
item = self.hf\_dataset\[idx]
\~\~\~\~\~\~\~\~\~\~\~\~\~\~\~^^^^^
File "/home/qiaojun/flexiv\_pi0-dev/.venv/lib/python3.11/site-packages/datasets/arrow\_dataset.py", line 2780, in **getitem**
return self.\_getitem(key)
^^^^^^^^^^^^^^^^^^
File "/home/qiaojun/flexiv\_pi0-dev/.venv/lib/python3.11/site-packages/datasets/arrow\_dataset.py", line 2765, in \_getitem
formatted\_output = format\_table(
^^^^^^^^^^^^^
File "/home/qiaojun/flexiv\_pi0-dev/.venv/lib/python3.11/site-packages/datasets/formatting/formatting.py", line 639, in format\_table
return formatter(pa\_table, query\_type=query\_type)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/qiaojun/flexiv\_pi0-dev/.venv/lib/python3.11/site-packages/datasets/formatting/formatting.py", line 403, in **call**
return self.format\_row(pa\_table)
^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/qiaojun/flexiv\_pi0-dev/.venv/lib/python3.11/site-packages/datasets/formatting/formatting.py", line 492, in format\_row
formatted\_batch = self.format\_batch(pa\_table)
^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/qiaojun/flexiv\_pi0-dev/.venv/lib/python3.11/site-packages/datasets/formatting/formatting.py", line 522, in format\_batch
return self.transform(batch)
^^^^^^^^^^^^^^^^^^^^^
File "/home/qiaojun/flexiv\_pi0-dev/.venv/lib/python3.11/site-packages/lerobot/common/datasets/utils.py", line 204, in hf\_transform\_to\_torch
items\_dict\[key] = \[torch.tensor(x) for x in items\_dict\[key]]
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/qiaojun/flexiv\_pi0-dev/.venv/lib/python3.11/site-packages/lerobot/common/datasets/utils.py", line 204, in <listcomp>
items\_dict\[key] = \[torch.tensor(x) for x in items\_dict\[key]]
^^^^^^^^^^^^^^^
RuntimeError: Could not infer dtype of dict

Exception ignored in atexit callback: \<function \_exit\_function at 0x7f0d4945db20>
Traceback (most recent call last):
File "/home/qiaojun/.local/share/uv/python/cpython-3.11.11-linux-x86\_64-gnu/lib/python3.11/multiprocessing/util.py", line 360, in \_exit\_function
p.join()
File "/home/qiaojun/.local/share/uv/python/cpython-3.11.11-linux-x86\_64-gnu/lib/python3.11/multiprocessing/process.py", line 149, in join
res = self.\_popen.wait(timeout)
^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/qiaojun/.local/share/uv/python/cpython-3.11.11-linux-x86\_64-gnu/lib/python3.11/multiprocessing/popen\_fork.py", line 43, in wait
return self.poll(os.WNOHANG if timeout == 0.0 else 0)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/qiaojun/.local/share/uv/python/cpython-3.11.11-linux-x86\_64-gnu/lib/python3.11/multiprocessing/popen\_fork.py", line 27, in poll
pid, sts = os.waitpid(self.pid, flag)
^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/qiaojun/flexiv\_pi0-dev/.venv/lib/python3.11/site-packages/torch/utils/data/\_utils/signal\_handling.py", line 73, in handler
\_error\_if\_any\_worker\_fails()
RuntimeError: DataLoader worker (pid 2343650) is killed by signal: Terminated.

Tavish9, 2025-05-12:
cc @TangGuohh

hairuoliu1, 2025-05-13:
Is there any solution?

Tavish9, 2025-05-13:
Currently, the merging utils is only for video dataset. 

The author might be busy these days, and will enhance this feature in the near future.

Thanks for your understanding!

closed 2025-05-24 by hairuoliu1