Tavish9/any4lerobot #36: robomind2lerobot error

swstbecrpgmail, 2025-06-12:
### System Info

Thank you very much for open-sourcing this tool. However, when run "sh covert.sh" to convert robomind to lerobot, it reports error. May I ask if you know how to debug this error, or what the possible causes might be?

[image]

### Reproduction

export HDF5_USE_FILE_LOCKING=FALSE
export RAY_DEDUP_LOGS=0
export RAY_TMPDIR=/nas15/rayTmp1
python robomind_h5.py \
    --src-path /nas15/bala/data/dataset/data2process/robomind_decompress \
    --output-path /nas15/bala/data/dataset/data2process/robomind2lerobot \
    --benchmark benchmark1_0_release \
    --embodiments agilex_3rgb franka_1rgb franka_3rgb simulation sim_franka_3rgb sim_tienkung_1rgb franka_fr3_dual tienkung_gello_1rgb tienkung_prod1_gello_1rgb tienkung_xsens_1rgb ur_1rgb \
    --cpus-per-task 2

### Expected behavior

successfully converting the data

Tavish9, 2025-06-12:
Hi, please run `python robomind_h5.py --help` for available arguments.

`python robomind_h5.py --debug` would be helpful.

swstbecrpgmail, 2025-06-12:
> Hi, please run `python robomind_h5.py --help` for available arguments.
> 
> `python robomind_h5.py --debug` would be helpful.

Thank you very much!
I have confirmed that the code error is related to the LeroBot version. In the latest version of LeroBot, the module lerobot.common.robot_devices is not present. Additionally, there are still errors when processing a few trajectories. I may attempt to resolve these issues in the future.

swstbecrpgmail, 2025-07-04:
> Hi [@swstbecrpgmail](https://github.com/swstbecrpgmail), could you please share the dataset that has dimension inconsistency?

Hi, as far as I know, "benchmark1-open_cap_trash_can, benchmark1-place_in_block, benchmark1-push_across_push_away_basket" have different image dimensions across their trajectories (but the image dimensions within each trajectory are consistent). In practice, I commented out the error message to skip the dimension check, allowing the code to run normally for data conversion. Then, I manually resized the converted videos to a uniform dimension and modified the info.json file.

Again, thank you very much for open-sourcing this convenient tool and you early reply!

Tavish9, 2025-07-04:
> Hi, as far as I know, "benchmark1-open_cap_trash_can, benchmark1-place_in_block, benchmark1-push_across_push_away_basket" have different image dimensions across their trajectories (but the image [...]

It's a known issue.

Please refer to [official explanation](https://github.com/x-humanoid-robomind/x-humanoid-robomind.github.io?tab=readme-ov-file#%EF%B8%8F-hdf5-file-format-%EF%B8%8F) for the inconsistency.

I recommend skipping `benchmark1_0_release/franka_3rgb` when converting.

Tavish9, 2025-07-04:
> Besides, the language instructions and images in the file "benchmark1_0_release/h5_tienkung_xsens_1rgb_pour_bread_then_place.tar" do not align, and I have not verified the other datasets yet.

If you want to report a bug, please provide a **detailed reproduction** steps, including any **relevant error messages**. Thank you! 😊

Tavish9, 2025-07-04:
> It is not a bug; it is a mistake in the original RoboMind dataset and is not caused by the conversion process.

The misalignment is also known, check the [README.md](https://github.com/Tavish9/any4lerobot/tree/main/robomind2lerobot#%EF%B8%8F-dirty-tasks) carefully.

If you want to go through the whole datasets, welcome for the pull request and post all the misalignment in readme, thanks.

YangYang-DLUT, 2025-12-02:
File "/mnt/dolphinfs/ssd_pool/docker/user/hadoop-mlm-hl/hadoop-mlm/yangyang260/proj/any4lerobot/robomind2lerobot/robomind_h5.py", line 239, in save_as_lerobot_dataset
    if config["images"]["camera_top"]["shape"] == (720, 1280, 3):
KeyError: 'camera_top'
/mnt/dolphinfs/ssd_pool/docker/user/hadoop-mlm-hl/hadoop-mlm/common/embodied_data/RoBoMIND/benchmark1_1_release/h5_agilex_3rgb/28_packtable_2/success_episodes/train, exception details: [36mray::save_as_lerobot_dataset()[39m (pid=114838, ip=10.245.136.77)
  File "/mnt/dolphinfs/ssd_pool/docker/user/hadoop-mlm-hl/hadoop-mlm/yangyang260/proj/any4lerobot/robomind2lerobot/robomind_h5.py", line 146, in save_episode
    ep_metadata = self._save_episode_data(episode_buffer)
  File "/mnt/dolphinfs/ssd_pool/docker/user/hadoop-mlm-hl/hadoop-mlm/yangyang260/proj/lerobot/src/lerobot/datasets/lerobot_dataset.py", line 1376, in _save_episode_data
    if not self.writer:
AttributeError: 'RoboMINDDataset' object has no attribute 'writer'

How to solve this?